keirf/flashfloppy-osd #2: A1200 instructions

penfold42, 2019-09-21:
Do you want me to provide them ?

keirf, 2019-09-21:
Yes please, if you could do it and take photos that would be especially great! Hopefully most signals have a convenient pad or component that does not require super skills to solder to. eg. keyboard test points (Rev 2), or fdailing that the keyboard pullup resistors. CSYNC and Green must be available on some lower-pin-count ICs or passives in the video circuit.

penfold42, 2019-09-21:
I’ve already hacked a working version onto my 1200 1D1.

I have my own keyboard intercept project in development and I use a PLCC socket on top of the CIA for KB data and clock.
This is probably the easiest way for boards without the keyboard test pins as the pull-ups are underneath.

Video, for now is on the DB23 as I use s-video

keirf, 2019-09-21:
Okay that makes sense. Those will be contact points that work generically for any SMT Amiga I would imagine. I also think it would be plausible to solder fine wire direct to the CIA. We don't need a picture of that necessarily. I may mod one of my machines and take a pic of that.

penfold42, 2019-09-22:
I cant see how to edit the wiki so pictures below:

A1200 rev 1.D1 video connections:
E260R - red
E259R - green
E258R - blue
E253R - Csync

[image]

The keyboard is a bit more of a challenge - the pullups are on the rear.
Here is a pic showing the pins on an upturned PLCC socket 
[image]

Also is a pic of my actual video wiring:
[image]

keirf, 2019-09-22:
Thanks! The wiki is not public editable. I don't know if I can add you but anyway I will add these details in myself not a problem.

keirf, 2019-09-22:
Would it have been better to connect to the output side of the 200R ferrite beads?

penfold42, 2019-09-22:
They’re not 200 Ω
I just measured them and they’re 0.2 Ω

I suspect they’re 200 μH

keirf, 2019-09-22:
Yes sorry. You're probably right or it's 200 ohm impedance to high freq AC. Either way your hook up is obviously fine. It's all up on the wiki as if this morning. Thanks!

closed 2019-09-22 by keirf