keirf/flashfloppy-osd #3: SSD1306 OLED emulation ?

penfold42, 2019-09-21:
Is this under consideration for a future release ?

keirf, 2019-09-21:
Yes this is interesting though there are two reasons I went for LCD in the first place:

1. I can pick a suitable OSD font. 8x8 fonts seem to work well, the taller fonts used for OLED displays may not look so good (but I haven't tried them).

2. I2C DMA transfers seem to cause occasional glitches in the OSD. I think this is because the start time of the Display-Output DMA is critical and it can be delayed if an I2C DMA transfer has already just begun. OLED I2C is run by FlashFloppy at 4x the LCD data rate. This may case 4x the glitches, and that may be unacceptable. Again, I haven't tried it to actually find out.

It's something to try as an experiment in due course and see how it goes.

keirf, 2019-09-21:
Some thoughts on issue 2. Perhaps I2C DMA can be paused for a short time (eg <10us) at the critical point on each scan line. eg. Set up a TIM2 or TIM3 channel in output-compare mode, and enable IRQ on compare-match. This could IRQ eg 2-3us before the OSD box, where we disable I2C DMA. Now we can either spin for eg 6us, or retask TIM1 to interrupt us again.

Former is easier but note that CPU activity can also jitter the OSD (bus-matrix contention presumably).

One possible solution: After disabling DMA, reprogram TIM1, then trigger a slightly lower-pri IRQ. This IRQ handler can WFI until TIM1 fires.

This is all subject to confirmation that I2C DMA can be safely paused at all. I have noticed when trying this before that subsequent DMA transfers are corrupted, but possibly disabled DMA for too long.

penfold42, 2019-09-21:
Item 1 - you could double the image in the X direction as an option ?

keirf, 2019-09-21:
Issue 2 is now completely fixed:
1. Now use IRQ rather than DMA for the I2C transfers. This is resilient to transfers being slightly delayed.
2. High-priority IRQ (higher priority than I2C IRQ) triggers 1us before the OSD box and holds the CPU for 1-2us while the critical DMAs are triggered.

keirf, 2019-09-26:
FF OSD now works alongside an existing OLED quite happily at 400kHz rate. Driving bitmap data to FF OSD at that rate would not be a problem.