keirf/flashfloppy-osd #4: Add RGB support rather than mono

penfold42, 2019-09-21:
Is the DMA engine fast enough enough to DMA bytes to a port ?

You then only need a handful of resistors to do 2 or 3 bit per channel colour.

For bonus points, I think some amigos have a blanking signal that could be abused to pre darken the image while the OSD is on

keirf, 2019-09-21:
Not fast enough to do it at current resolution. The best would be about half the current resolution (more like how hxc OSD looks) and that's dma at full rate so timing of every dma is critical or the OSD jitters. Still it could be tried with say 6 pins in the b8-15 range.

keirf, 2019-09-21:
That's assuming 32 bit dma transfers are as fast as 16 bit transfers. I expect they are.

penfold42, 2019-09-22:
Maybe we could borrow the pixel driving code from:
https://github.com/abelykh0/VGA-demo-on-bluepill/blob/master/README.md

Then you could possibly use the SPI bus as a slave to capture keyboard data without missing an edge

keirf, 2019-09-22:
Yes amiga keyboard input could be grabbed via spi. I wouldn't be able to filter out my hot keys and hide them from the host, which is a nice touch.

I guess during display output that project sits in a loop which must not be interrupted or contended by anything (not even systick, let alone i2c irqs or dma)

penfold42, 2019-09-22:
I forgot about your bit 7 hack

Lucky it’s all open drain and the right polarity!

keirf, 2019-09-22:
Yes it was fortuitous 😅

keirf, 2019-10-09:
I think this is a no. OSD will remain monochrome.

closed 2019-10-09 by keirf

penfold42, 2019-10-09:
All cool - the display enable lets me drive the 3 channels to get better contrast which was my primary aim.

I hope to have some hardware designs on github by end of this weekend