keirf/flashfloppy-osd #5: A600 video info

penfold42, 2019-09-22:
If you are only using RGB analog (and not composite) the same approach as the A1200 should work.
E253R	CSYNC
E260R	Red
E259R	Green
E258R	Blue

If you also want the OSD to work via composite (or S-Video if you have modified it) then we will need to test a slightly different approach. If will likely involve injecting the video into R212 or R215:
[image]

keirf, 2019-09-23:
I will add info on this to the wiki. Looks like no handy test point for kbd signals either, sadly.

keirf, 2019-09-26:
I have updated the wiki, please take a look. I'm pretty sure mixing OSD into the signal at R211-213 will work fine. Perhaps the 150-ohm impedance will require a slightly bigger-than-usual resistor in the OSD wire.

penfold42, 2019-09-26:
Looks good - I may soon have access to a 600 for validation

keirf, 2019-09-26:
It would be nice to get photos too... Perhaps we leave this open for a bit.

penfold42, 2019-09-26:
Of course !

closed 2019-12-09 by keirf