keirf/flashfloppy-osd #9: Simplify gotek wiring

penfold42, 2019-09-28:
Could you emulate a second pcf8574 with FF-OSD and feed Amiga keys  to it ?

Could you add support to FlashFloppy to support buttons on a pcf8574 ?

If so, we can eliminate 2 or 3 wires between them and everything is done over i2c.

As a bonus, you could add support for more buttons on the gotek
- up down left right.
- a shift key for page up/down

keirf, 2019-09-28:
Since we don't emulate a pcf8574 when talking to latest FlashFloppy, the protocol can be extended to be bidirectional. We'd probably want to version the protocol at the same time, and continue to handle already-installed OSDs which don't support bidirectional transfer over I2C. That can't be hard to detect.

So yes, I suppose then I'd need to hook up the new 'buttons' to do something. And I'm not sure the extra three wires at the moment is so much of a pain. The FF-OSD board will probably getsintegrated into an enhanced Gotek, or glued on the back of a standard one, anyway: So the wiring disappears or is static.

keirf, 2019-10-06:
I have a proof of concept going for this. So from next version of FF OSD I will disable Gotek button pins when in extended protocol mode, and send the button info via i2c instead. This will also require a new version of Flashfloppy of course, which will precede the next version of FF OSD.

keirf, 2019-10-08:
I spent some time implementing this only to realise at the very end that having physical connection to the buttons is very useful for triggering firmware update. The bootloader makes a very early decision whether to execute or update based on reading GPIOs.

The only ways round I can see are:
1. You would only need one button wire, for SELECT/JA. That suffices to enter the updater.
2. Could add a menu option in the main firmware, to run the updater. This could work in a few ways:
 * Erase first page of firmware, then reset. Bootloader detects this already.
 * Write to the backup registers (preserved across reset) and then reset. Modify bootloader to read the backup registers.
 * Jump at the bootloader direct. Possibly harder than it seems since the chip state is not reset, as the bootloader can currently assume.

penfold42, 2019-10-08:
FWIW - and I’m a stm newbie “Write to the backup registers“ seems the cleanest.

You could also have the FF validate/verify that the USB stick has a valid UPD file.

keirf, 2019-10-09:
This is all done as of FF OSD v1.7 and FlashFloppy v3.5a.

The old A3-A5 connections are no longer supported and must be removed.

Latest FF does not work with older FF OSD due to I2C changes.

closed 2019-10-09 by keirf