keirf/flashfloppy-osd #12: Hot keys should be edge triggered

penfold42, 2019-10-01:
I haven’t put the scope on it yet but holding multiple keys looks like it’s doing PWM on the outputs

keirf, 2019-10-01:
Oh I meant to fix that. But all the same given the BSRR write should match the existing pins settings, I'm surprised the output pins oscillate?

keirf, 2019-10-01:
I can't repro unstable output pins, is that due to other changes in your tree?

keirf, 2019-10-01:
Oh hang on I see what you mean: Multiple hot keys simultaneously? Why would you do that...

closed 2019-10-01 by keirf

penfold42, 2019-10-01:
Never underestimate the stupidity, or creativity, of explorative testers

keirf, 2019-10-01:
Haha, now fixed anyway :) Maybe not worth a new release in itself, as requires user to build the firmware themselves anyway, to see the problem.

My current minor annoyance is my first display of the OSD box is glitching again. That means the first time after power on, or OSD reset, that the OSD box appears (eg. Ctrl-Alt-Return) I see an instantaneous screen glitch. It does not happen again until next full reinitialisation. Can you repro that? It's very weird and I have no idea why it happens!

commit 32e05d26 (2019-10-01): Amiga: Hotkeys are edge triggered (ie. only on initial key press)
Fixes #12