keirf/flashfloppy-osd #14: Support for BBC Micro/Master decoded keyboard

dandoore, 2019-10-27:
Is it possible to assign pins on the “Pill” to utilise the output from the BBC micro keyboard (and from the Master’s compatible PL7 header) to allow keyboard control?

keirf, 2019-10-27:
Is the keyboard something that's easy to intercept? I think it's not a serial protocol like Amiga and atari st?

dandoore, 2019-10-27:
I’m not overly sure but schematic wise it looks like an 8 bit value returned in parallel on a clock cycle when another pin is asserted, presumably as part of the keyboard scan routine.

dandoore, 2019-10-28:
There’s a description of how it works here: 
http://www.cyberspice.org.uk/blog/2011/04/27/the-bbc-model-b-microcomputer-keyboard-how-it-works/ and the schematic for the Master here (which shows both the Master onboard decode custom chip attached to headers PL11 and PL24 and the BBC B header at PL7) http://mdfs.net/Info/Comp/BBC/Circuits/Master/master2.gif
also the keyboard matrix http://mdfs.net/Info/Comp/BBC/Circuits/BBC/bbckbd.gif if that helps!

penfold42, 2019-10-29:
That looks pretty messy for the OSD to also do.
 Could you use an atmega to sniff the BBC and then clock it out to look like Amiga key presses ?

keirf, 2020-01-17:
I don't think this is in scope and, since Beeb doesn't have a side-facing drive like Amiga/ST, it seems low priority too.

closed 2020-01-17 by keirf