keirf/flashfloppy-osd #15: VGA timing ?

penfold42, 2019-10-31:
Is this hard coded to 15kHz ?

How hard and is it and is the SPI bus fast enough to do 31kHz timing ?

Use case is Amiga and ECS inidivision type devices

keirf, 2019-10-31:
It is not hardcoded. The SPI bus is only running at a quarter of max speed. For 31KHz timing I would run it at 18MHz rather than 9MHz.

penfold42, 2019-11-14:
I’ve made some progress at https://github.com/penfold42/FF_OSD/tree/my_board?files=1

The timing is a a compile time option for now, but I’ll do some auto detection.

penfold42, 2019-11-16:
Here’s a screenshot at 18MHz pixel clock

[image]

If I over clock the APB1 to 72MHz so I can get a 36 pixel clock it looks even better -

[image]

This is obviously naughty - the cleaner option if we want this is to move to SPI1 which is on ABP2 and officially  supports running at 72 to get a 36MHz pixel clock

keirf, 2019-11-16:
No reason not to use MOSI1 on PA7 really. The lack of 5v tolerance doesn't matter for this signal.

Have you tried it?

penfold42, 2019-11-16:
Not yet - I’ve been looking at measuring the csync line rate to try vga/15kHz auto detection.

I’ve already got the dynamic change working tied to Amiga got keys for now

Firstaid2008, 2019-11-23:
Hi. Sorry for being a little bit of a novice.  I am wanting to use FlashFloppy OSD on an indivision ECS.
I have done all the flashing of the blue pill. Soldered all the wires GREEN HSYNC VSYNC to the output from the ECS. I have checked the monitor and it says the HSYNC is 39khz.  SO how do i change the setting to get this to work.  Thanks all.

penfold42, 2019-11-23:
Did you grab the code from my fork and the “my_board” branch ?

1st thing is to configure sync polarity - negative is I think the default and it needs to be positive (or vice versa)
Without an OSD, that’s a bit hard ! But the menu is available also n the serial port.

Do you have the Amiga keyboard lines connected as well ?
If you do, you can change the sync polarity and timing with the key combinations instantly

penfold42, 2019-11-23:
Oh and btw, the vga support is just as novice as you are at the moment!
Also, don’t go making PCBs just yet - some pins might change
And another thought - my branch assumes an external video buffer so if you don’t want that there’s a small code change needed

Firstaid2008, 2019-11-24:
So i need to build an extra board to be able to send to vga ???

penfold42, 2019-11-24:
> So i need to build an extra board to be able to send to vga ???

Are you referring to the green pcb at the top of the shot ?

No - the Indivision isn’t mine and I didn’t have a spare IDC socket to crimp onto the vga cable.

I would either solder wires to the back of the idc connector on the Indivision or crimp on another 10 pin IDC socket to the ribbon cable

Firstaid2008, 2019-11-24:
I have just got a reply from Keir Fraser. And he has put me straight on what i need to do.  Only thing is i have no knowledge of coding. So i dont know how or what way to compile the code you have.  I have read the code and i think i should be changing the 15khz to 37khz.  Then compiling and flashing this to the blue pill.  Is this correst.

penfold42, 2019-11-24:
No changes needed if you grab from the vga branch I just pushed:
https://github.com/penfold42/FF_OSD/tree/vga?files=1

If you aren’t planning on developing for OSD I can post a hex file for you if you like

Firstaid2008, 2019-11-24:
yes please. That would be terrific.  Then all i would have to do is flash that to the blue pill and i should be good to go ???

penfold42, 2019-11-24:
Yep!

Gimme 10 mins

Firstaid2008, 2019-11-24:
Thankyou Very Very much.

penfold42, 2019-11-24:
https://raw.githubusercontent.com/penfold42/FF_OSD/vga_test/penfold42/FF_OSD.hex

penfold42, 2019-11-24:
Keen to hear how you get on - I suspect you’re only the second person to try this (other than me!)

Firstaid2008, 2019-11-24:
Not very well.  Still no joy.  I have connected B15 to pin 2 of vga out of indivision with 270ohm inline. A8 to pin 13 of vga. B14 to pin 14 of vga. Pin B3 to E631. B4 to E632.   GND from gotek to G on blue pill.
3V from gotek to 3.3 on blue pill.   SDA from gotek to B7. SCL from gotek to B6. There is one 4.7kohm resistor from the 3.3V to both SDA and SCL.
[image]
[image]

  The Blue Pill lights up.
If i connect RXD from serial to A9 and TXD to A10. I can flash the blue pill. And i have checked this by Clearing the blue pill.  Flashing the hex file you sent me. Then changing jumper and running putty. Press reset and i get FF OSD v1.7 **   Current config: Sync Active HIGH. etc.   To check it had flashed correctly.  I again cleared blue pill and flashed the original from Keir Fraser. and the info back from Putty is different.  So cleared it again and again flashed your FF-OSD.hex.  But still no joy cant get anything up on the monitor. Which if i goto settings on the monitor AOC 21inch.  It says HFrequency 39khz. V.Frquency 62khz.  But there is nothing on screen from Blue pill.

keirf, 2019-11-24:
You need two 4.7k resistors: one for SCL and one for SDA. You have tied the lines together and it's no wonder it doesn't work!

Now I look at the wiki page it could be clearer so I'll let you off. ;)

Firstaid2008, 2019-11-24:
yes there is 2 4.7k resistors one for each. Sorry if i put it wrong
[image]
The grey is the 3.3v White is SDA and purple is SCL.

keirf, 2019-11-24:
Ah yep that's okay :) So when you bring the OSD config up on serial line, you don't see the config settings up on the host screen? That is the first thing to get working as it doesn't depend on working connection to the Gotek at all. All it really needs is the SYNC + GREEN signals.

Firstaid2008, 2019-11-24:
Yes i get config up on serial line via putty. Press reset and i get

** FF OSD v1.7 **
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FF_OSD

current config:
 Sync: Active LOW
 H.Off: 42
V.Off: 50
Rows: 2
Columns: 16-40

Keys:
 Space: Select
 O: Down
 P: Up
Sync Lost
line_rate 0 line_count 0

Does that look right ??

Firstaid2008, 2019-11-24:
This is using Penfold42 compile.

keirf, 2019-11-24:
Looks fine. If you press space on serial console you should enter configuration menu, and *that* should make something appear on your host video screen.

Firstaid2008, 2019-11-24:
I have also now put a jumper across A0 - A1 and connected an OLED screen and the screen is working fine.

Firstaid2008, 2019-11-24:
yes it goes to config. And if you press space at each they all come up one after the other.

FF OSD v1.7 Flash Config 
Sync: Active LOW
H.Off (1-199):  39
V.Off (2-299): 62
Save New Config? Save

I can change the settings as instructions    O/P to change   and Space to continue.

keirf, 2019-11-24:
Ok will have to wait to see what penfold42 has to say.

Firstaid2008, 2019-11-24:
Sorry guys my bad. I Re soldered all connections on blue pill.  I now have OSD.   Need sync to HIGH.   and it needs VSYNC or it wont work.

penfold42, 2019-11-24:
Woohoo !

Sorry for not highlighting the Vsync requirement -will get it documented in the wiki when my code gets merged.

Do you know if sync polarity was High after you first flashed it ?

Just a heads up: you may need to move one wire in the future - i want to get double the resolution so the OSD box isn’t as wide.

Firstaid2008, 2019-11-25:
I think it was set to Active LOW.  And thankyou again.

Firstaid2008, 2019-11-30:
Sorry Penfold42.  I have just flashed another blue pill for my A600 with indivision ecs and when i flashed. It reports to default Sync High.

penfold42, 2019-11-30:
Excellent ! Thanks for the update.

I haven’t had much time lately to look at moving things around for even better vga resolution.

Maybe I should make a pull request for possible inclusion as it stands now ?

keirf, 2019-11-30:
I asked for a PR for the timer swap?

penfold42, 2019-11-30:
I know - it wasn’t a 10 min job and I got busy.

I’ve spent ages getting my head around the timers, dma and interrupts
The Amiga keyboard is also using timer 3 and I don’t think this can be changed.

keirf, 2019-12-01:
OK no problem. The use of tim3 by the amiga code is not a problem except it uses the tim3 irq. And the timer.c code would want to do same. They just need to share a common irq entry point which demuxes to their respective handlers based on interrupt reason (bits in TIM SR register).

keirf, 2019-12-01:
I can sketch a patch for this if you like.

penfold42, 2019-12-01:
I’m getting there - it doesn’t help that I have a killer cold !
Work so far if you want a peek 
https://github.com/keirf/FF_OSD/compare/master...penfold42:vga_spi1

Amiga keyboard now works again
It finds sync, can still drive the flashfloppy, just no video output !
I suspect I’m still misseing a timer change or two

keirf, 2019-12-01:
Comments:
1. TIM4 clock is no longer getting switched on. Perhaps move all the RCC_APBxENR_TIMxEN into stm32f10x.c:peripheral_init() -- it's probably less confusing!
2. The test-and-clear of TIM4->SR flags is racy. Remove "tim->sr = 0" from IRQ_timer(). Instead put "tim3->sr = ~my_sr" in my_IRQ_29(): This will clear only the flags you collected in your read of tim3->sr.

keirf, 2019-12-01:
I would like the timer switch separate and tested on existing 15kHz setup with existing display-out pin assignment if possible, please.

penfold42, 2019-12-01:
Yep - I haven’t forgotten - I need to test it with 15k and vga so for now doing the development with the vga stuff.
When happy, you’ll have a clean PR with 1 commit for the timer swap.

Thanks for the tips above - it might be time for me to get my head around GDB if I I’ve missed other things

penfold42, 2019-12-07:
#22

keirf, 2019-12-09:
I have applied all your patches so far. Very little modification except to the final patch, which had a bunch of unnecessary global vars.

However, I have the following follow-up comments:
1. config.display_autosync should be part of the config.display_timing enumeration rather than a separate config field and option. This will mean that a new variable will be needed in main.c to track current actual display_timing.
2. I don't much like the new Amiga keys. This is all already config'able in the config menu, where it properly belongs. I suppose hotkeys are handy for your development of course: But now they should be removed or #if 0'ed out at least.

Can you pick these up?

penfold42, 2019-12-09:
Thanks!

Yep - I’ll do the enumeration.

On the hot keys - you’re right about the value during development.
For daily use tho, I thought they might be useful if you shoot yourself in the foot for quick changes when you can’t see the menu.

keirf, 2019-12-09:
I figure users are more likely to shoot themselves in the foot with these hotkeys. At least if they screw up configuration, they were in the configuration menu, and their bugreport is likely to say so. Only the WASD keys I'm more lukewarm on, but still I think config should be config.

penfold42, 2019-12-11:
If you’ve got time/interest I’ve done the 
 screen mode menu with auto as an added option.
I’ve also added auto sync polarity detection and added in the same way (high, low, auto)

Are there some “#ifdef debug” set in the build process I can use ?

keirf, 2019-12-11:
Do we need polarity detect? It really is pretty much always LOW for 15k and HIGH for VGA? Well if it's reliable I suppose it doesn't hurt :)

Yes happy to see patches.

There is no debug=y hookup. Look in FlashFloppy for how that works: Pretty much it's picked up by a few lines in Rules.mk and sets -DNDEBUG if debug is not 'y'.

penfold42, 2019-12-11:
It’s 100% reliable ... on my two test cases being my A500’s ECS indivision VGA and A520 modulator outputs !
Jokes aside, I think the C128’s CGA output is 15kHz positive sync so I’ll test that too

penfold42, 2019-12-12:
#25

keirf, 2019-12-12:
Well it looks fine to me so I merged it!

I'm not sure the xxx_MAX values are correct: In my book the max is the max valid value, not +1. But that's the sort of thing someone can clean up later.

penfold42, 2019-12-12:
Great, thanks!

"someone" - the owner of more action and backlog items in the world.
If I have another kid im calling them "someone"

BTW, I've just finished updating the wiki pages - as soon as i can work out how to push it to my fork ill let you know here...

penfold42, 2019-12-12:
It helps if you actually create the wiki on a fork.... done
 https://github.com/penfold42/FF_OSD.wiki

keirf, 2019-12-12:
I merged your Wiki changes and cleaned them up. Cheers.

keirf, 2019-12-12:
Anything more for this ticket?

penfold42, 2019-12-12:
Nope !

closed 2019-12-12 by penfold42