keirf/flashfloppy-osd #16: USB HID Keyboard support

cmaxwellau, 2019-11-07:
Support a USB HID keyboard on the OTG port and generate kb_data

keirf, 2019-11-07:
STM32F103 doesn't have OTG.

cmaxwellau, 2019-11-07:
Im using a pair of 105s on a custom gotek/ffosd board

keirf, 2019-11-07:
Then yes you can. It's out of scope for FF-OSD though. My interest is seamless integration into pristine existing hardware.

cmaxwellau, 2019-11-08:
No problem. Luckily the 103 and 105 in 44 pin variety have virtually the
same footprint, and the price difference is negligible. My aim is to
similary maintain a pristine interior of my A600 & A1200 without all the
mess of bluepills, ribbons and 3d printed bracketry with a small footprint
gotek based entirely on your Hardware Mods list. Thanks for all the
fantastic work you've done on FF and OSD.(just to name a few)

keirf, 2019-12-09:
By the way I have a plan next year to port FlashFloppy and FF_OSD into an integrated build targeting STM32F730R8. Could be interesting. Those chips have dual OTG.

cmaxwellau, 2019-12-10:
Haven't had a look at those chips yet.

I've just gotten my first prototype boards for Gotek+FFOSD with 2x105s.
Happy to share/open up the kicad repo if there's interest. It also includes
CIA adaptor boards for THT/SMD sockets.

keirf, 2019-12-10:
I'm happy to put designs up in the wiki.

tkurbad, 2020-02-08:
@cmaxwellau I'd be very interested in your combined Gotek/OSD design as I was planning to do something similar (based on OpenFlops). My aim is to design a board that fits into a 5,25" bay alongside a real 3,5" floppy, while maintaining all of FF's goodies, like rotary, speaker, and OSD.

closed 2020-11-10 by keirf