keirf/flashfloppy-osd #18: OSD does not show ADF

Primy-games, 2019-11-18:
Hello good day.
I have installed OSD on an A500 Rev 5 and apparently everything works.
L. Ctrl + L. Alt + Return works.
L. Ctrl + L. Alt + Help works.
L. Ctrl + L. Alt + Del works.
L. Ctrl + L. Alt + Left Up and Right do not work. It does not show anything on the screen I can not switch between the different ADFs, but if the GOTEK buttons work, however I can configure the OSD menu.
what can happen?

keirf, 2019-11-18:
Do you have a jumper installed across A0-A1 and are you using latest FlashFloppy v3.x?

Primy-games, 2019-11-18:
Yes, the jumper is in A0-A1 and the firmware is version 3.7A, this version is programmed with USB-TTL to avoid failures.

keirf, 2019-11-18:
Do the arrow keys actually work? Have you tested them on the Amiga?

Primy-games, 2019-11-22:
I have done several tests with the keyboard and it works perfectly, I have recorded a video with the error if there is a way to indicate where the problem is. Thanks greetings.

https://youtu.be/pqOCq13rmkY

keirf, 2019-11-22:
And so we're clear on the situation, what is the FF OSD version number?

keirf, 2019-11-22:
Ah I see in the video... Latest v1.7? You don't have any wires installed from OSD to gotek buttons right (you shouldn't)?

keirf, 2019-11-22:
I don't see any output from the gotek to the OSD. Are you sure the gotek is talking to the osd? When you change image it should appear on screen, for example.

Primy-games, 2019-11-25:
The version of the OSD is the latest 1.7, I have no cable connected to the GOTECK buttons, and nothing appears on the screen when switching between ADF, it seems that effectively as you say you do not communicate the Goteck with the OSD. The diagram used for the connection has been as follows:

    SCL = B6
    SDA = B7
    VCC = 3.3
    GND = G

I'm going to try changing the lines:
SCL = B7
    SDA = B6

and I comment the result, since the goteck is also flashed with the FF 3.7a version with USB-TTL.

keirf, 2019-11-25:
Did you attach a 4.7k pullup resistor from each of B6 and B7 to 3.3 ? It won't work otherwise.

Primy-games, 2019-11-28:
I hope you can apologize to me, do not connect the 3.3v to B6 and B7, for that reason there was no communication. Thank you very much for your work, a big greeting

closed 2019-12-09 by keirf