keirf/flashfloppy-osd #19: Moving display to SPI1 ?

penfold42, 2019-11-24:
I’ve started trying to get SPI1 working so we can get higher resolution - primarily for VGA style screens.

The gpio and SPI #defines are obvious
I’ve also enable the ABP2 clock for it.

Where I’m stuck is DMA - it looks like it uses DMA3 which is also used by one of the timers.

Can they share it ?
Or is it easier to move timer3 to another timer ?

Thanks!

keirf, 2019-11-24:
Swap timer3 with timer4 (currently used in timer.c, should only need to change the couple of lines at the top of the file to change timer and its irq number). 

Please send this to me in a self-contained pull request after testing at 15kHz too, and it can go straight in the main tree.

keirf, 2019-11-24:
Bear in mind in the ongoing SPI work that 15kHz needs to stay at B15 as there is existing hardware to support. So the SPI output will basically have to be preferably run-time selectable, perhaps as simple as 15 vs 30kHz operation.

keirf, 2019-11-24:
Or perhaps we move to new pin based on a jumper setting... hmmm.... We'll have to give it a bit of thought.

keirf, 2019-12-02:
Above PR is manually merged. Please note that a significant modification was to drop to TIMER_IRQ_PRI when doing timer.c work. Please let me know if this merged patch breaks your world!

penfold42, 2019-12-02:
Thanks !

Will do

penfold42, 2019-12-03:
My world is good - just finished re-implementing my work on top of this

Next step is making it run time switchable.
Do you think we should have independent control of pixel clock and SPI output ?
or require that 15kHz /9MHz uses SPI2 and vga/36MHz uses SPI1 ?

keirf, 2019-12-03:
Keep them independent I reckon. Default SPI2. Then you can get a reasonable if slightly wide output from VGA.

By the way if I port FF to F730 chip and integrate OSD functionality, we will have different SPI baud rates to choose from: power-of-two divisors of 108MHz. So obvious ones will be 6.25, 13.5, 27.0. (54MHz would probably be too fast even for VGA?).

13.5MHz would likely be good for 15kHz mode, which is a bit lo-res still at the moment.
I'm thinking 27.0MHz will be okay for VGA? It's perhaps a smidge narrow at 36MHz.

penfold42, 2019-12-03:
That would be cool.

Another option is to put the SPI into slave mode and use an external clock to clock the pixels out.
The tick then is phase locking it to the hsync pulse to eliminate jitter.

penfold42, 2019-12-05:
Getting close now
https://github.com/keirf/FF_OSD/compare/master...penfold42:vganew?expand=1

The only thing that doesn’t work is dynamic SPI pin switching immediately - a save and reset works fine.

keirf, 2019-12-05:
I don't think that matters. I'd rather require save&reset if the
alternative is a new cumbersome interface into the core timer stuff.

penfold42, 2019-12-07:
#22

keirf, 2019-12-09:
Unfortunately the F730 has only 64kB Flash so the QSPI (actually only DSPI on the base 730R8 part) will be needed for QSPI Flash for main firmware :)

keirf, 2019-12-09:
Closing this and keep discussion on #15

closed 2019-12-09 by keirf