keirf/flashfloppy-osd #27: OLED display dropping out & changing config with Gotek buttons

bradwh, 2020-02-02:
Two issues and maybe related.
First Issue 
After a fresh flash from the FF upd file both my OSD and OLED display work fine and remain working fine until the system is power cycled. After a power cycle the OLED display remains blank. No amount of navigation, button pressing will bring the display back again and behaves exactly the same no matter how many power cycles. However a fresh flashing of the FF firmware from the .upd file on the USB and up comes the display again and will remain working correctly until next power cycle. 
Running the GoteK without the FF_OSD Bluepill SCL & SDA lines connected then the display works fine through multiple power cycles fine.
Currently running with latest version firmware for both FF v3.11 and FF_OSD v1.8 however I have tried older versions of both with the same behaviour. I have tried both with and without the 4.7K pullup resistors on the SCL & SDA lines with no difference in behavior. Of course A0 & A1 on the Bluepill are jumpered to enable the dual Display.

Second issue.
FF_OSD configuration access. This is not attached to an Amiga so the Amiga keyboard option is not available. I can successfully enter and configure the FF_OSD configuration using the serial method through pins A9 & A10 of the Bluepill, successfully changing and customising the on screen FF display. However I just can't get into it from the Gotek buttons. I can eject an image and I can navigate through the eject menu with either my rotary encoder or Gotek buttons I.E. copy image, delete image etc, but I can't find an option "Configure FF OSD" to select. 

At the moment I'm waiting on a delivery of a couple of fresh Bluepills as I'm not 100% confident of this one. Although I have confirmed that the crystal is running at 8MHz the Blink runs at 1Hz not 2Hz but all other aspects seem to be fine, communication at 115200 baud works fine so maybe an anomaly of some kind but maybe a contributing factor to the first issue above but unlikely to be causing my problems with the config access.
Any and all suggestions welcome.

keirf, 2020-02-02:
Are you getting a three row display on the OSD with one row double height?

keirf, 2020-02-02:
The wiki was out of date regarding entering the Config Menu from FlashFloppy. This is now fixed.

FlashFloppy now has a main menu, entered by pressing both buttons (or your rotary encoder) when no USB stick is inserted. *Configure FF OSD* has moved to this main menu.

bradwh, 2020-02-03:
Thanks Keirf,
Yes I do get a three row display on the OSD with one double height, this is despite the fact that I have the OLED display type is currently set as oled-128x32 and the display order is set to OLED 128x32: 0,1.

I have confirmed that I can now enter the Configure FF OSD setup with the when no USB stick is inserted. Now to get the buttons configured to enable select.

keirf, 2020-02-03:
Okay here is a FlashFloppy fix. Please give it a go!

[ff_osdfix_27.zip](https://github.com/keirf/FF_OSD/files/4146684/ff_osdfix_27.zip)

bradwh, 2020-02-03:
Thanks Keirf, 
That looks like it has solved the problem with the OLED display. I can now go through multiple power cycles without the OLED failing to operate as it should.
I still can't seem to select or navigate the Configre FF OSD which I can now see when powering up without the USB inserted, using the Gotek's left / right buttons or the rotary switch.
In addition the changes to the ff_osdfix firmware appears to have changed the operation of the OSD itself. I no longer get any of the data that is being shown on the OLED display showing up on the OSD window. However if I enter the OSD Configure with the serial connection all the data that shows on the PUTTY session is displayed onto a two line version of the OSD until I save or save - reset, then the window disappears never to return until I bring up the Serial OSD config again with a putty session.
Do I need to flash a new version of the Blupills firmware to match the changes in the Gotek that you sent above?

keirf, 2020-02-03:
Okay, for selecting *Configure FF OSD*: You should be able to press the rotary switch twice. Once to select. Once to confirm. And then the display should say *Exit: Power Off* and all further button/rotary activity is forwarded to FF OSD.

Of course it might not appear to be working because it seems OSD is now broken :)

You don't need to update FF OSD, v1.8 is fine. It's possible my patch is just broken, or that somehow FlashFloppy now detects OLED but somehow fails to detect FF OSD at the same time. That latter seems unlikely. I'll stare at the patch and maybe test it myself...

keirf, 2020-02-03:
Oh I know why: *Configure FF OSD* is unselectable because FlashFloppy has not detected your OSD.

What we don't know is why your OSD is no longer detected. I assume nothing at all changed hardware-wise. OSD was working, then you updated, then OSD immediately no longer worked?

I may need to add some debug logging and get you to run the logfile firmware...

keirf, 2020-02-03:
Here is a debug firmware, please run the logfile firmware (https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive). You should just need to power on with USB stick inserted, observe the problem, and immediately pull the stick and attach the FFLOG.TXT file to your reply here.

[ff_osdfix_27_2.zip](https://github.com/keirf/FF_OSD/files/4148875/ff_osdfix_27_2.zip)

bradwh, 2020-02-04:
Thanks Keirf,
No hardware changes other than a change of USB drive (using one for the Original v3.11a firmware and one for the v3.11a oled fix firmware).
I can confirm we have a full reversal of that was happening before. The OSD now works straight after a fresh flash of the Gotek and subsequent power cycles cause it to remain blank just like we were seeing with the OLED display on the old software. The OLED remains working fine. So it looks like the problem with power up initialisation has swapped from the OLED to the OSD Bluepill with the change in Firmware version.
Attached is the FFLOG.TXT so hopefully you can get some helpful information from that.
[FFLOG.TXT](https://github.com/keirf/FF_OSD/files/4151895/FFLOG.TXT)

keirf, 2020-02-04:
Hi, Sorry that's the wrong build! I think you are running ff_osdfix_27.zip in the logfile above.

The correct build timestamp (printed in FFLOG.TXT) should be 16:17:56.

Interestingly though I do see that OLED and OSD appear to both get detected on first run (immediately after update?) and that after that indeed OSD never gets seen again... Anyway, I added an extra debug line in ff_osdfix_27_2 and that's what I'd like to see :)

bradwh, 2020-02-04:
Thanks Keirf, I’m away from home for a few days will look again on my return. Thanks for your patience.
Regards Brad

keirf, 2020-02-05:
I reproduced this issue myself, and worked around it. So please try the following firmware (no need for logfiles, just update to the normal firmware and see if your OSD and OLED both reliably spring to life!):

[ff_osdfix_27_3.zip](https://github.com/keirf/FF_OSD/files/4158765/ff_osdfix_27_3.zip)

bradwh, 2020-02-05:
Thanks Keirf, will be back home tomorrow night, so give it a go🤞
Regards Brad

bradwh, 2020-02-07:
Thanks for that, so far looks good, flashed the new version and both displays came up OK.
Power cycled and both displays came on, let it time out and then changed image and both displays came up ok.
Run at least ten more power cycles and both displays came up OK each time. Also confirmed the correct operation of navigating and selecting the OSD configuration via the Gotek and rotary encoder buttons OK.
Looks like we have a winner.

keirf, 2020-02-07:
Okay this fix will be in next FF release, v3.12a. Meanwhile you can use your private fixed version.

closed 2020-02-07 by keirf