keirf/flashfloppy-osd #29: Documentation? How to enable Amiga keyboard bindings for Gotek?

SvOlli, 2020-08-16:
Hello!

I've just finished my first build of an A500 with a FlashFloppy+OSD. Everything is running fine, except for that the OSD controller does not send the keyboard events from the Amiga to the Gotek. Amiga Keyboard works for configuration and Gotek keys work for selecting floppy images displaying info on OSD. Only the keyboard commands intended for the Gotek simply don't work. It doesn't matter if I go for keyboard hold or use CTRL + ALT + CRSRs.

To me it seems like I need to configure something either inside the OSD, the FF.CFG or both, reading through the docs for some time now, I didn't find anything. Or doesn't it just work, because the I2C is implemented as a "one way" communication?

keirf, 2020-08-16:
Did you put a jumper across A0 and A1 on the osd blue pill board?

SvOlli, 2020-08-16:
No. As I understood the docs, this is only needed when OSD is used alongside OLED.
Nevertheless, I tried it, and now the keyboard works, but when data from the Gotek is displayed, the screen turns purple from the point to the bottom of the screen, full width, without any text displayed.

keirf, 2020-08-16:
But it worked better than that (the actual display on screen) without the jumper?

The jumper is really needed as it enables the extended bidirectional protocol between gotek and OSD.

SvOlli, 2020-08-16:
Yes. The two cases are:
jumper off: display good, no keyboard
jumper on: display broken, keyboard works
I'm assuming switching to bidirectional somehow causes ff to send other commands / commands differently to the blue-pill osd. Maybe my setting of "display-type = auto" doesn't work?

keirf, 2020-08-16:
Did you reconfigure OSD at all, eg reposition the osd box? The gotek will default to a 40 char wide OSD box when the jumper is set. If you repositioned the osd box it could overlap end of the display line with unpredictable results.

keirf, 2020-08-16:
BTW you can jumper A1 to A2 at power on to reset the osd config. I had to look in the code to remember that. I must improve and simplify the wiki!

SvOlli, 2020-08-16:
Yes, indeed I did, and I shortened it to 20 chars and moved it to bottom right. Now I can see something and only need to figure out the best setting for display-type as about 40% of the right-hand space is purple but unused.

keirf, 2020-08-16:
Yes you can adjust width with lcd-xxx settings. Leaving it wide will mean that long names won't need to scroll. But it will leave a lot of blank space for short names.

keirf, 2020-08-16:
I will leave this ticket open until I have time to do some work on the wiki.

SvOlli, 2020-08-16:
I'm also still running into smaller problems.
- the filenames are truncated to 24 chars, leaving the rest unused.
- configuring display-type=lcd-40x04 or lcd-24x04 will just turn off any display (except for config)
- setting "display-order=3,1,0,2" still leaves me with the default configuration

SvOlli, 2020-08-16:
But that's for another day. Do you have a recommended version of FF.CFG for use with a OSD-only A500 configuration?

keirf, 2020-08-16:
Looking back at the code, you can't change the OSD box dimensions or display row ordering. It will always be 40x3 when A0-A1 is jumpered. If you set display-type other than auto and no LCD/OLED is attached, display setup will fail. This really needs to be fixed with a separate set of config parameters for OSD: dimensions, row ordering, and auto-off parameters.

As for filenames truncated to 24 characters, that shouldn't happen. Are you direct navigating using the OSD, or are you using the file selector method (AUTOBOOT.HFE)?

SvOlli, 2020-08-17:
I'm using direct navigating. When the 7 segment display is replaced by either an OLED/LCD or OSD it's much more convenient to use.

keirf, 2020-08-17:
So what happens? Are the names truncated to 24 chars no scrolling? Or do the full length names scroll within a 24-char area, despite the OSD box being 40 char?

I may have to test it myself.

keirf, 2020-08-17:
Hmmm agreed I can reproduce this. How weird. I will have to investigate.

keirf, 2020-08-17:
Okay, this has been broken in FlashFloppy since forever, embarrassingly enough. Try this fixed firmware please: It should use all 40 characters for very long filenames. (Note this is for the Gotek, not the OSD hardware!)

[ff_osd_29_1.zip](https://github.com/keirf/FF_OSD/files/5085283/ff_osd_29_1.zip)

SvOlli, 2020-08-17:
Yes. This now works as expected. Thanks!

chapas, 2020-11-07:
Hi, having the same issue as described by @SvOlli, A0-A1 jumper ON or OFF yields the same result, keyboard interactions doesn't work. I'm with a A500 rev8A, kickstart v2.0.
What else can I try?

Thanks

keirf, 2020-11-07:
Does CTRL-A-A reset work on your Amiga?

chapas, 2020-11-07:
Yes, it works.

keirf, 2020-11-07:
As I followed up / edited to say: download amiga test kit and test your whole keyboard

SvOlli, 2020-11-08:
@chapas : I also did some tinkering with the FF.CFG, As I was trying quite a couple of things, I can't say which one got it working for me. Here's my FF.CFG with the comments removed, maybe it helps.
```
interface = shugart
host = unspecified
pin02 = auto
pin34 = auto
write-protect = no
side-select-glitch-filter = 0
track-change = instant
write-drain = instant
index-suppression = yes
head-settle-ms = 12
motor-delay = ignore
chgrst = step
ejected-on-startup = yes
image-on-startup = last
display-probe-ms = 3000
autoselect-file-secs = 3
autoselect-folder-secs = 3
folder-sort = always
sort-priority = folders
nav-mode = default
nav-loop = yes
twobutton-action = eject
rotary = full
indexed-prefix = "DSKA"
display-type = default
oled-font = 6x13
oled-contrast = 143
display-order = 3,2,1
display-off-secs = 15
display-on-activity = no
display-scroll-rate = 200
display-scroll-pause = 2000
nav-scroll-rate = 80
nav-scroll-pause = 300
step-volume = 20
da-report-version = ""
extend-image = yes
```
This worked on my system with an FF 3.15.

chapas, 2020-11-08:
Thanks @keirf and @SvOlli.
Tested with 2 different keyboards, none of them worked, except for the reset?!? :(
Changed to another Amiga500 rev6, both keyboards fully working (every single key) tested using Amiga Test Kit.
Back to the Amiga500 Rev8A, swapped CIA ICs (don't remember where I saw this a long time ago) and now both keyboards are working!?!?!?

Now on a new problem.
I can change disks using keyboard, but don't see the change happening on screen.
To try to fix something ended up messing the configuration of the FF_OSD, now can't even see the configuration, but can 'blindly' change disks. A full reset to FF_OSD with a key combination might be a suggestion.
Back to square zero, will try to re-flash FF_OSD to erase the messed up configurations, but right now I don't remember which bootloader I programmed nor how I programmed FF_OSD in the end (Alzheimer maybe).
Which bootloader should I re-flash on the blue pill in order to have it enabled for programming the same way (DFU mode) as the Gotek.

SvOlli, 2020-11-08:
> none of them worked, except for the reset?!?

The reset is on a separate pin on the A500. So if there's a "malfunction" in either the clock or the data connection, no data would get transmitted, except for the reset. The A2000 keyboard, for example, differs in this part.

keirf, 2020-11-08:
You can perform a factory reset by placing a jumper at A1-A2 then powering on. You can safely use the A0-A1 jumper for this purpose. Then switch off, put the jumper back at A0-A1, power on, and you should be back to working.

chapas, 2020-11-08:
Thanks @keirf, hadn't solder a pin on A2, but that is clear on the documentation. As easy as it was, a key combination was easier :P
Got it to work without reprogramming anything.
Thanks for all the help.

keirf, 2020-11-09:
A0-A1 requirement is clarified in the wiki now.

closed 2020-11-09 by keirf