keirf/flashfloppy-osd #31: Length of I2C Bus wires

ILAHWWINC, 2020-10-25:
I have my FF-Gotek in a external case connected to my Amiga 2000. The FF-OSD BluePill is inside the A2000. In the FF-Gotek there is a OLED connected parallel. 

The connection of the I2C Bus is over a cable, 50cm, pluggable. This setup works sometimes, but not reliable.

May the length if the cable be the problem? Any ideas to make it working reliable?

keirf, 2020-10-25:
Pull-up resistors in the i2c display may be too high for the wire capacitance. You could try 1k resistors soldered from each of SDA and SCL, both connecting to VCC.

keirf, 2020-10-25:
Sorry not in the i2c display, in your FF-OSD. Try replacing the 4.7k resistors, or double them up (to halve the effective resistance).

bradwh, 2020-10-27:
I can confirm that it can work fine over that length of I2C.
I have a similar arangement running over a 1.8 Meter (M-M PS/2) sheilded cable from the FF-GOTEK to my FF-OSD also with an OLED fitted in parrallel inside the FF-GOTEK. I'm also running a rotary encoder over seperate conductors of the same cable.
In my case it worked fine with out any modifications just the 4.7K pullups at the FF-OSD end.

ILAHWWINC, 2020-10-28:
I can confirm that reducing the
pull-up resistor to 2k make it work here.
Thanks for helping me.

closed 2020-10-28 by ILAHWWINC