keirf/flashfloppy-osd #32: Amiga keyboard control issue

BwanaFr, 2020-11-09:
Hello,

I mounted a FF OSD (1.8) with a gotek (FF 3.21) in an Amiga 500 (keyboard CLK and DATA are connected to the Blue Pill).
Vidéo is working as expected and I can activate/hide OSD using Amiga keyboard and enter into FF_OSD configuration using keyboard).
But I can’t control Flashfloppy using arrows keys on Amiga keyboard.
As the Blue pill displays properly the FF selected disk, I guess the I2C communication is ok.
The Gotek has no configuration file and I reset the flash configuration.

Any ideas of why I can’t control the Gotek using Amiga keyboard?

Many thanks!

keirf, 2020-11-09:
Place a jumper at pins A0-A1 on your blue pill.

closed 2020-11-09 by keirf

BwanaFr, 2020-11-11:
Jumper was already in place but I suspect a bad contact.
Anyway, it works very well now!

Many thanks Keir!