keirf/flashfloppy-osd #36: Inverted Mode using OSD

jgilcas, 2021-01-05:
I have installed a BP in a spectrum and the result has been great, but there is a detail that I do not understand, everyone appears the letters in color and to me appears the background color and transparent letters, I have reviewed configuration but can not find anything. What can be? I did not have 4k7 and I put 10k, to pull up would work the same, right?
[image]

keirf, 2021-01-05:
Have you attached to the green signal only in the Spectrum? Perhaps you are overdriving the signal, in which case you should try connecting with a bigger series resistor to the green signal.

jgilcas, 2021-01-05:
Maybe i´m wrong and my picture is ok. Let me attach two pictures.

Non Inverted?
[image]

Inverted?
[image]

keirf, 2021-01-05:
Sorry yes, I think all the above are correct including your original picture. Assuming you are on the green signal, lettering will be adding green and the OSD box background is subtracting green.

On the Spectrum, the white background is already "full green". You can't add more green. Hence the lettering appears like the main background (white), while the OSD background is predominantly red+blue (=purple).

On the Kickstart screen, the white background is a dull white and the analog green can be overdriven to give a pale green. Again the OSD box is predominantly red+blue=purple, but it just happens to be paler here.

And the Amiga Test Kit screen is predominantly dark red, not much blue or green. So the OSD text stands out as vivid red and the OSD box, having little of any RGB, is almost black.

So, that's just how it is :)

closed 2021-01-05 by keirf

penfold42, 2021-01-05:
The only way to make it “better” is to use an analog switch and completely replace the original video signals.
There’s a display enable signal that can be used to drive this but it’s a much more complicated install as you need to more electronics and you need to cut traces on the computer.

jgilcas, 2021-01-05:
No no, this solutions is perfect for me, my mistake was not try with
another backgrounds, now i tested with black background and it's perfect,
awesome.