keirf/flashfloppy-osd #39: Cant get a synced video signal on a Amiga 500 Rev 8A

ninjafahrer, 2021-02-24:
FF OSD is correctly wired.
I am using Firmware 1.8 together with FF Geotk 3.16A.
I am getting no sync with the video Signal, have a lot of green lines on the screen.
The Amiga is connected via Scart to a TV with a PAL signal.

FF OSD settings are on Auto.

keirf, 2021-02-24:
Do you have a picture of the effect? Do you have a ground connection to the Amiga board (via I2C ground wire via Gotek will do, so probably this is ok)? Do you have a picture of your connections to the Amiga mainboard?

ninjafahrer, 2021-02-24:
[
[image]
[image]
[image]

keirf, 2021-02-24:
Where exactly do you pick up _CSYNC signal? It's hard to tell where that second wire goes, which appears to terminate directly below the ODD CIA lettering. But I'm pretty sure there's no _CSYNC in that little area.

ninjafahrer, 2021-02-26:
[image]

penfold42, 2021-02-26:
I think thats the wrong pin for CSYNC
try this:
[image]

https://www.amigapcb.org/ is a BRILLIANT site

keirf, 2021-02-26:
As @penfold42 says.

ninjafahrer, 2021-02-26:
Great, that solved the problem!

Thanks for your support!

closed 2021-02-26 by keirf

penfold42, 2021-02-26:
Excellent !