keirf/flashfloppy-osd #40: Can't get custom build for hotkeys to work

solarmon, 2021-04-25:
Hi,

I'm trying to test the custom hotkeys feature to work - as per the example at:

https://github.com/keirf/FF_OSD/blob/master/src/default_config.c

I have changed `#if 0` to `#if 1` to try to test the first example - see my fork of this project at:

https://github.com/solarmon/FF_OSD/blob/master/src/default_config.c

I have also changed the version to 1.8.1 in the Makefile just to prove that it is a new build:

https://github.com/solarmon/FF_OSD/blob/master/Makefile 

I'm building it in Ubuntu in a WSL environment in Windows 10. The build compiles OK and I am able to firmware flash the .hex file to the Blue Pill.

However, when I try the key combo of CTRL+LALT and the Function keys, I do not see any corresponding text appear on OSD, or the OLED display. The FF OSD is working OK, as the OSD display works and it is mirroring the dual connected OLED display correctly. This is also connected to an RGBtoHDMI setup, but I don't see that being relevant, or a potential cause of the issue?

What other troubleshooting and debugging would you recommend? Ultimately, I would like to get the hotkeys feature to work with RGBtoHDMI, but this is just an initial test to prove that the hotkeys feature is working.

Once I can get it to work then I can create a suitable default_config.c example to support the the three buttons on RGBtoHDMI.

Thank you.

keirf, 2021-04-25:
Tested your build and it works for me. Are other OSD Amiga key combos working for you?

solarmon, 2021-04-25:
@keirf Thanks for checking.

Yes, I can use the OSD Amiga key combos - I can control the Gotek, select between images, etc; also access and control the FF OSD menu.

I'm testing this on rev 5 A500, if that makes any difference, and also with my RGBtoHDMI board with FF OSD support (tri-state buffer) that can display FF OSD on RGB and HDMI.

I can see the FF OSD version as 1.8.1, so I know I have flashed my custom build. Are there any debugs that log the key combos detected?

keirf, 2021-04-25:
I'm testing on a Rev 6A A500, though revision and model shouldn't matter.

There is no logging but if you do a debug build you will get serial-log output, and you could add printk() logging to update_amiga_keys() in src/main.c. 

One thing to add might be:
```
for (i = 0; i < 10; i++) {
    if (amiga_key_pressed(AMI_F(i+1)))
      printk("[F%d]", i+1);
}
```

And do a debug build and then see if you get lots of `[Fn]` when you press C-A-Fn.

solarmon, 2021-04-25:
@keirf 

I did the debug build and the serial output does catch the function keys combo presses:

[image]

But still nothing being shown on OSD - either on RGB or over HDMI.

I have the display timer on and OSD is not displayed after a period of time. I noticed that even though the function key combo is detected then, the OSD display does not appear (like when another key combo is detected).

I'm testing this now on my rev 6A A500. I'll try to test also without the RGBtoHDMI integration.

keirf, 2021-04-25:
Welcome to debugging. You probably want to add more printk() to the hotkey checking loop in update_amiga_keys(). See how it filters out the hotkey options based on:
1. Is there a hotkey set up for this Fn key?
2. Has this Fn key changed state?
3. Has the key just become pressed?
And then sets up the `notify` structure which gets picked up around line 1069 in src/main.c when rendering the OSD.

solarmon, 2021-04-25:
@keirf

So I tested FF OSD just on it's own, without RGBtoHDMI, and it was still having the same issue.

So I added some test output in the debug code for the function key detection - I used the same code as for turning OSD OFF or ON (Del key):

`
/* ### DEBUG */
	for (i = 0; i < 10; i++) {
    if (amiga_key_pressed(AMI_F(i+1))) {
		printk("[F%d]", i+1);
		
		snprintf((char *)notify.text[0], sizeof(notify.text[0]),
                 "TEST F%d", i+1);
        notify.cols = strlen((char *)notify.text[0]);
        notify.rows = 1;
        notify.on = TRUE;
        notify_time = time_now();
		
		}
	}
	/* DEBUG ### */
`

And this works - I see "TEST F1" for when I do the CTRL+LALT+F1 key combo. Similarly, for the other function keys I see the corresponding test text for it.

solarmon, 2021-04-25:
@keirf 

In my debugging, I'm outputting the following at the start of the `update_amiga_keys()` function:

```
/* ### DEBUG */
printk("[ARRAY_SIZE: %d]", ARRAY_SIZE(config.hotkey));
/* DEBUG ### */
```

This always seems to come back with a value of 10.

Shouldn't it be what is defined in `default_config.c` - i.e. however many hotkeys that were defined?

keirf, 2021-04-25:
The array is always 10 entries long. It's defined so in Inc/config.h and is indexed by Fn key.

solarmon, 2021-04-26:
OK, thanks, I read that wrong then.

The next question. What should this piece of code do? Should it only jump to the next iteration of the loop if the function key is not defined? If I add debugs to this bit of code I get output for all 10 loop iterations.

```
if (hk->pin_mod == 0)
            continue;
```

If I add debugs to these next sections of the code, I don't get anything at all, so it seems the key presses are not being matched here.

```
/* Has hotkey press/release state changed? */
        hk_pressed = amiga_key_pressed(AMI_F(i+1));
        if (!((hk_latch>>i & 1) ^ hk_pressed))
			/* ### DEBUG */
				printk("[PRESSED 1 F%d]", i+1);
			/* DEBUG ### */
            continue;
        /* State has changed: Is the hotkey now pressed? */
        hk_latch ^= 1u << i;
        if (!hk_pressed)
			/* ### DEBUG */
				printk("[PRESSED 2 F%d]", i+1);
			/* DEBUG ### */
            continue;
        /* Hotkey is now pressed: Perform configured action. */
        s = (uint16_t)hk->pin_high << pin_u0;
        r = (uint16_t)(hk->pin_mod & ~hk->pin_high) << pin_u0;
        gpio_user->bsrr = ((uint32_t)r << 16) | s;
        if (*(p = hk->str)) {
			/* ### DEBUG */
				printk("[PRESSED 3 F%d]", i+1);
			/* DEBUG ### */
            notify.cols = notify.rows = 0;
            memset(notify.text, 0, sizeof(notify.text));
            while (*p) {
                int len = strlen(p);
                strcpy((char *)notify.text[notify.rows], p);
                notify.cols = max(notify.cols, len);
                notify.rows++;
                p += len + 1;
            }
            notify.on = TRUE;
            notify_time = time_now();
        }
```

keirf, 2021-04-26:
I wonder if you have an old flash config hanging around? Can you try reset to factory in the config menu?

keirf, 2021-04-26:
Or jumper A2-A3 (I think?) to clear flash config at power on

solarmon, 2021-04-26:
OK, so I was on a relatively old  FF v3.16.

I tried first factory resetting the config on FF, but it did not change anything. Then I updated it to v3.25, and it didn't change anything.

Then I factory reset FF OSD (using the FF OSD menu) and it started outputting the 'PRESSED 1' debugs.

However, I can't get it to show the 'PRESSED 2' or 'PRESSED 3' debugs yet, but at least something has changed.

keirf, 2021-04-26:
Sounds like progress. Yes indeed I meant clear FF OSD config not FF. :)

solarmon, 2021-04-26:
I had forgotten I had changed it to the other example, so hotkeys for F9 and F10.

You can see the 'PRESSED 1' debugs for F9 and F10 and when I press F9, it keeps outputting debugs for 'PRESSED 1 F9':

[image]

But there is no subsequent 'PRESSED 2' or 'PRESSED 3' debugs - so it looks like it is not matching on the button release?

keirf, 2021-04-26:
I would think with the old config cleared you should be able to do a clean build with your debugging removed now and see if it all works?

solarmon, 2021-04-26:
Even with the debugging it should be working anyways? I've only added debugging output to the test build.

But, yes, I'll try a clean build with hotkey support and see what happens.

solarmon, 2021-04-26:
OK, I build a clean firmware with the second example (F9/F10) and when I loaded I set FF OSD to factory reset and I saw the correct OSD text for when I pressed F9 or F10.

I built the firmware again, but this time used the first example (F1-F4, F9/F10). After loading the new firmware I did set FF OSD to factory default. I noticed that the previous F9/F10 example was still working. After factory defaulting FF OSD, the new example (F1-F4, F9/F10) started to work.

So my conclusion is that it looks like a factory reset of FF OSD is required for any new/different custom hotkey builds.

keirf, 2021-04-26:
Yes it's stored as part of the flash config (kind of thinking that perhaps this could be edited at runtime in future). But at the moment it just causes confusion like this :)

solarmon, 2021-04-26:
Maybe at a minimum, somehow show the current hotkey config in the OSD Menu?

keirf, 2021-04-26:
It probably makes sense in config_init() to unconditionally copy `dfl_config.hotkey[]` to `config.hotkey[]`

keirf, 2021-04-26:
Okay, see the fix in branch `issue_40` and whether that fixes all the confusion for you.

solarmon, 2021-04-26:
I tried the issue_40 branch and it does indeed allow me to change builds with different hotkeys without having to factory resetting FF OSD.

commit b367dd0d (2021-04-26): Always use the compile-time hotkey config.
Fixed #40
closed 2021-04-26 by keirf