keirf/flashfloppy-osd #48: Usage of FlashFloppy-OSD in MSX2 computers

Wierzbowsky, 2022-07-31:
First of all - this is an awesome project and I see quite a few use cases for it in the MSX world. In some cases Goteks have to be put in places where they are not visible, so this project comes in very handy! I will start from the slim Gotek and Al-Alamiah AX-350 MSX2 computer and continue with the normal size Gotek and Yamaha YIS-503III converted to MSX2+.

Are there any concerns that I should note when connecting STM32 board to MSX computer? I am going to connect it first to a computer with v9938 and then to the computer with v9959 VDP chips. Both of them have CSYNC and RGB outputs. I presume that it should work like on Amigas - without extra tweaking, but you never know.

Also, it is possible to get 3 outputs from the STM32 board so that all 3 color signals (R+G+B) could be patched and the OSD would be white? In some cases having monochrome output based on one color component would not work.

Connecting to MSX computer's keyboards could be problematic, but technically possible. @keirf could you please check the keyboard connection on this schematics and tell whether it's possible to patch into the keyboard to use SHIFT, CRTL and arrow key combinations?
https://sysadminmosaic.ru/_media/msx/yamaha_yis-805-128r2/yis805.pdf

keirf, 2022-07-31:
You should be able to hook into msx no problem. To connect to all of rgb you might want to hook the stm rgb output via three tristate buffers and hook each of those to your rgb lines. OSD can be configured to control tristate buffer output enable.

Keyboard will be hard. It is amiga specific and uses the fact amiga keyboard is smart and uses a serial protocol. Msx seems to have an 8 bit style keyboard where matrix scanning is done by main host cpu.

Wierzbowsky, 2022-08-01:
Thanks, I will try it later this week. I am a bit familiar with Arduino, so why to use a 3-state buffer when one can configure 2 other outputs of the "Blue Pill" board to do the same as B15 does? Then we have 3 lines doing the same thing and each one can be connected to its own color channel.

keirf, 2022-08-01:
Timing is tight and not enough spare timers.

Wierzbowsky, 2022-08-02:
I've got my STM boards, but they appear to be STM32F103C6T6A, not STM32F103C8T6. Connected everything, flashed the latest OSD firmware, but nothing works. Will the current FF OSD firmware work on these C6 boards? I presume C6 is a slightly different chip...

Also, would it be possible to light the greed led in the firmware when it OSD is on? This could help in debugging...

keirf, 2022-08-02:
C6 has less Flash and SRAM. Actually I think it could work except the flash config is saved beyond the end of C6 flash capacity. Maybe a one line change would help...

keirf, 2022-08-02:
Give this version a go (random version number change so you can track which version you've flashed).

[FF_OSD-v1.33.zip](https://github.com/keirf/flashfloppy-osd/files/9244875/FF_OSD-v1.33.zip)

Try to connect over serial or using a rotary encoder as described in the wiki to enter configuration mode. This will display OSD box even if your I2C connection to Gotek isn't working yet.

Wierzbowsky, 2022-08-02:
Thanks. But still no image via MSX's RGB. I connected the board to PC via serial IO and got this output:

[image]

Is the OSD output enabled by default?

keirf, 2022-08-02:
Should be. Press space over serial to enter config and see what happens.

Wierzbowsky, 2022-08-02:
It shows config entries every time I send "Space", but I can't change anything. And eventually the device stops responding. I connected the B15 via 220 Ohm resistor (could not find 270) to the R input of the video module (see the schematics). The A8 is connected to the SYNC IN input of the video module.

keirf, 2022-08-02:
Does it show the config entries on the OSD though?

Wierzbowsky, 2022-08-02:
Nope, the screen remains empty

keirf, 2022-08-02:
You will probably need to put a scope on the pill output pin to look for signs of life, and take it from there. I think on serial you should see logging when csync is locked on, so absence of that may mean some csync issue.

Wierzbowsky, 2022-08-02:
I will do that. What should be the amplitude of the acceptable CSYNC? Now I connect it to the input of the video module, where it is already TTL level.

Wierzbowsky, 2022-08-02:
Tried to play around with settings. When I select something else for "Display Enable", the red color disappears from the video. Other changes make no effect on the picture. OSD still not visible. But looks like the Flash config works.

[image]

keirf, 2022-08-03:
Maybe but that would be surprising. What is the horizontal frequency of the msx output?

Wierzbowsky, 2022-08-03:
15.7kHz, vertical can be 50 or 60hz depending on model

Wierzbowsky, 2022-08-03:
No problem, the C8 board is on the way. I will update you when I test with it.

Wierzbowsky, 2022-08-06:
I've got the C8 chip, it appears to have 128kb flash, unlike the one shown in [this video](https://www.youtube.com/watch?v=9noYsB_QJkQ). This should not be a problem though. Flashing v1.9 firmware worked fine.

[image]
[image]

When I connect the SYNC/A8 to the LS04 chip that brings the CSYNC to TTL levels, there comes the message "sync found" and there's no sync loss until I disconnect the wire. But the OSD still doesn't work. B15 is connected to the input RED signal pin of the video module via 220 Ohm resistor. The serial interface works:

[image]

Btw, there's no "sync found" message on the board's power-up, only when I disconnect and connect the SYNC wire. Probably it would be good to report the sync finding on startup.

Wierzbowsky, 2022-08-08:
I've made a few experiments with Gotek and Blue Pill and monitored it with the scope. The CSYNC signal that is fed to STM board via A8 looks like this:

[image]
[image]

The RED signal that is connected to B15 without the output from Gotek looks like that:

[image]

There's no activity on B6/B7 - I2C bus when the STM board is connected to Gotek. When I connect the OLED display to Gotek, the screen works and there's expected activity on B6/B7. Should the screen be connected for the STM board to work? Maybe the STM board doesn't work with I2C but when it is directly connected?

The jumper A0/A1 is installed. The pull-up resistors are installed. The setup looks like this:

[image]

- Connected

keirf, 2022-08-08:
Do you have any cached display-type= setting in the Gotek FF.CFG?

Wierzbowsky, 2022-08-08:
> Do you have any cached display-type= setting in the Gotek FF.CFG?

This is my FF.CFG, works on all my modded MSXs with Gotek.

[FF.zip](https://github.com/keirf/flashfloppy-osd/files/9284411/FF.zip)

keirf, 2022-08-08:
I can't read it, I'm afk this week. Does it have a display-type line?

Wierzbowsky, 2022-08-08:
I changed the display type to auto and now there's activity on I2C bus. SCL is 100kHz, SDA is alive too. But still no output on B15.

Wierzbowsky, 2022-08-08:
The output on B15 while the wire is disconnected looks like this:

[image]

keirf, 2022-08-08:
It's supposed to be tristate outside the box and then push-pull high-low with maybe a 3v swing inside the box. What does the display config look like as printed via serial?

Wierzbowsky, 2022-08-08:
> It's supposed to be tristate outside the box and then push-pull high-low with maybe a 3v swing inside the box. What does the display config look like as printed via serial?

[image]

keirf, 2022-08-08:
Looks okay I think. Flash blinky on it from greaseweazle firmware repository and see what that says on serial

Wierzbowsky, 2022-08-08:
> Looks okay I think. Flash blinky on it from greaseweazle firmware repository and see what that says on serial

Sorry, I can't find a HEX file for blinky test. Where can I get it? Thanks.

keirf, 2022-08-08:
Download latest release from greaseweazle-firmware repository. Unzip and look under hex/alt iirc

Wierzbowsky, 2022-08-08:
Thanks. Found HEX, flashed it and... garbage on serial output on reset, LED doesn't blink. I am novice to this platform, so I may be doing something in a wrong way...

The same happens with C6 board. No blinking LED, garbled serial output on reset. I am using 115200 speed, 8 bit, even, stopbit1.

keirf, 2022-08-08:
No blinking led is bad. No serial is weird. I may have broken the build slightly and it's not currently at 115200 baud. Actually I think that's likely. Try 3000000 baud (three million).

Wierzbowsky, 2022-08-08:
[image]

keirf, 2022-08-08:
Fake and or broken

keirf, 2022-08-08:
I can sell compatible at32f403 chip but you will need ability to swap chip over and make some resistor adjustments

Wierzbowsky, 2022-08-08:
Could it be that the chip is different from what the program wants to see? In the above mentioned video I've seen the 64kb version of C8 working with OSD, but my chip is 128kb version.

keirf, 2022-08-08:
It's definitely a fake. Unsurprising since genuine simply isn't available. Not necessarily a problem as long as the fake works properly. Clearly this one doesnt

Wierzbowsky, 2022-08-08:
Can the firmware be adjusted to run on clones/fakes?

keirf, 2022-08-08:
They can be broken in various ways. Life is too short.

Wierzbowsky, 2022-08-08:
I want to bring this project into the MSX community, but I can't do this when people can't get hold of the properly working boards. Is there any way around this problem? Do you know any supplier of the working boards?

keirf, 2022-08-08:
No the source of recent working versions have been using at32f403 either on custom boards (eg zaxxon amigotek osd) or swapped onto blue pill boards to replace bogus chips

keirf, 2022-08-08:
I may be able to support other pill like boards if one type looks promising

Wierzbowsky, 2022-08-08:
Black Pill?
https://www.amazon.de/-/en/STM32F411-Development-STM32F411CEU6-MicroPython-Programming-black/dp/B09DPQN5VX/

Wierzbowsky, 2022-08-09:
I still wonder whether anything could be done with those chips where I2C doesn't work well. I doubt that the I2C doesn't work in those chips, it probably works, but with some adjustments. Maybe lowering the bus' frequency or changing the protocol to take into account those known issues would be possible? You are a really good engineer, maybe you could find the solution?

keirf, 2022-08-09:
I probably could but I'm not going to. How many of this particular fake are out there? How many other fake types might I have to support? What's the bang for buck for my efforts? 

Supporting a completely different pill type I'd at least somewhat more compelling if they are coming with genuine chips or good fakes.

The current situation of needing to use (in practice) at32f403 chips is not ideal since they are hard for people to source except through me or gotek factory in larger quantity. And they need to be soldered.

Wierzbowsky, 2022-08-09:
This is, of course, understandable. But as you can see, there's no good solution to use FF OSD in Goteks for now. According to your own statement, almost all available STM32 chips on the Chinese-made pill boards are fakes. And 98% of people won't be able to swap a fake STM32 chip with an Artery's themselves. So, unless you start selling the pill boards with the swapped chips yourself, we are out of options. At least until you choose to support a different, easily obtainable pill board. All they keys are in your hands now, so please help us. 

Quite a few MSX systems have Goteks in side slots or mounted sideways, so the screen is not visible. And FF OSD is the best solution for those systems. From my part, I will encourage people to donate to you when FF OSD project will be available for MSX systems. I've already donated for the FF once and I will donate for FF OSD too.

keirf, 2022-08-10:
Perhaps I should port to STM32F401 black pill board? These seem generally available for no more than a fake 103c8 blue pill.

Wierzbowsky, 2022-08-21:
Received the replacement board from the German company. Blinker still fails.

[image]

Waiting for the C8 board from 4 different Chinese suppliers. Why would I2C fail on all these boards? They can't be all defective.

Wierzbowsky, 2022-08-27:
Finally got 4 Blue Pill boards from China. Opened 3 of them. All of them pass the I2C tests, but all fail on Flash tests. One of them outputs ID, so FF OSD says that it's fake. But all can be programmed, FF OSD saves the config without a problem.

[image]
[image]

Those issues, however don't stop the boards from working. I finally managed to get a working OSD on MSX!

[image]

Now I need to identify the best way to locate Blue Pill inside Gotek (especially inside a slim one), best points to connect color signal and sync, as well as try to tune the OSD brightness by playing around with the resistor value on color channel's connection.

keirf, 2022-08-27:
Good news. These chips probably have only 64kb Flash but that's all osd requires.

Wierzbowsky, 2022-08-28:
Some more info after testing. At least 3 problems were identified with v1.9 of FF OSD on MSX:

1. By default the OSD display height is double, while in the configuration it's set as "Normal": 
[image]
[image]

But when the UART is connected and configuration is pulled via Putty with space bar, the OSD size becomes normal again:
[image]

2. When the connected color channel is not identical to the background's color, there are color alterations on both sides of the OSD:
[image]
[image]

3. When 50hz PAL is selected on the MSX2 machine, the screen periodically becomes black at certain intervals when OSD is active. With 60hz NTSC this doesn't happen:

https://user-images.githubusercontent.com/16538424/187086413-d1403cce-b0e2-4b5b-a65b-2bc6a3a1c7d8.mp4

Wierzbowsky, 2022-08-31:
@keirf , what could cause the enlarged size of the OSD in my tests? Flash problems?

keirf, 2022-08-31:
It isn't enlarged, the middle row is sent double height by FlashFloppy. If you want a more vanilla two row OSD display then remove jumper A0-A1 from the OSD board.

keirf, 2022-08-31:
The bleeding and loss of monitor sync is strange. Perhaps the series resistor is too small value (bleed)? Loss of sync I don't know it could be because the osd pixel clock is not synced to the host pixel clock?

Wierzbowsky, 2022-08-31:
Removing jumper A0-A1 helped with the OSD size problem, thanks! Probably worth explaining the need for this jumper in the doc, now it's only mentioned that it enables some advanced I2C protocol. How this affects the size was not mentioned.

I will play around with resistor value and will try different sync connection point. @keirf do you think that a shielded wires should be used for CSYNC and color signal? My wire is about 30cm as Gotek is far away from the video module.

keirf, 2022-08-31:
I'm no expert in analog electronics but a foot is quite long and the circuit may have poor grounding too.

Wierzbowsky, 2022-08-31:
Looks like it works well with RGB now. I installed the pots on both CSYNC and color channel wires and tuned up the colors. The best results I get when color pot is 330 Ohm and CSYNC pot is 200 Ohm. Here's the color test table
[image]

Installing the pot on CSYNC seem to have resolved the black screen problem on 50 Hz machine. Sadly, the composite video shows pretty noisy picture, but the OSD is still usable with that output.

Will test with a few monitors later this week.

keirf, 2022-09-01:
Unfortunately because the pixel clocks aren't synced it will be a bit noisy horizontally on lcd. It looks good on crt!

Jumper A0-A1 according to taste. Most people like the double height image name.

Wierzbowsky, 2022-09-01:
@keirf I have a couple of more questions:

1. What is the B15's output method? Does it pull the signal up or down to create OSD? I want to try to add 3 diodes to the B15's output and connect them to 3 color channels at the same time to get white OSD. Your answer will help to identify which way to connect those diodes

2. Is there a way to bring OSD configuration with Gotek's 2 buttons? Normally, people don't install rotary encoders and connecting UART every time to configure the OSD is too much hassle

keirf, 2022-09-01:
1. B15 drives signal up *and* down to define the box area and highlight text. Outside the OSD box it is tristate. You cannot drive the RGB channels using diodes. You will need to use a tristate buffer chip like 74LS367 or 74HCT367. Connect the display out to three tristate buffers, and you can configure FF OSD to output a display enable signal on pin A15, which you would connect to 367's OE pin(s).
2. Yes press rotary or both buttons while no stick inserted to enter Main Menu. Configure FF OSD is an option in that menu. Once entered FF OSD Config mode, you will need to power cycle to exit back to normal Gotek mode iirc.

Wierzbowsky, 2022-09-01:
Something like this?
[image]

Do I need resistors for every color channel with this setup?

[9 more replies]

closed 2022-09-05 by keirf