keirf/flashfloppy-osd #49: Control the size of OSD or file name via setting in FlashROM

Wierzbowsky, 2022-09-05:
Currently, when the A0-A1 jumper is set, the OSD shows enlarged file name with a slash on top of the window:
[image]

This is not controlled by any setting stored in the FlashROM. The only way to disable this is to remove the jumper. But then the enhanced protocol stops working and it's no longer possible to configure FF OSD settings using Gotek's pushbuttons.

I propose to disable enlarging the disk's name using the jumper and instead enhance the options of the "Display Height" setting. There could be the following options: "Normal", "Double size" and "Enlarged disk name". Then people could keep the A0-A1 jumper and configure the OSD with Gotek's buttons without having the forcefully enlarged OSD frame. Much better than to connect UART all the time when there's a need to change something.

Wierzbowsky, 2022-09-07:
Thanks! I tested this version today. The slash in the upper left corner is gone, but the line it was at is still there, so there's the same gap on top of the OSD as before. 

By default, with A0-A1 jumper and no USB flash, the OSD comes with the enlarged DSK image name (this is expected):
[image]

With the FF.OSD with the proposed "osd-display-order" setting, the OSD is vertically normal, but still enlarged horizontally (not expected):
[image]

Would it be possible to fix the horizontal size too? Thanks in advance.

keirf, 2022-09-07:
I'll have a think about the horizontal size. Apart from that it's as expected. Banner text layout doesn't have configurable layout.

keirf, 2022-09-07:
The wide box can fit long filenames though. But you'd rather scroll?

Wierzbowsky, 2022-09-07:
> The wide box can fit long filenames though. But you'd rather scroll?

Yes, I'd rather have a compact OSD. The biggest benefit of the compact OSD is that you can put it in any corner of the screen.

keirf, 2022-09-14:
Do you want a special build? That's easy. Working out how to make it configurable would need more thought.

Wierzbowsky, 2022-09-14:
No, I'd better wait for the generic solution. However, if you disable the default enlargement of OSD (both horizontal and vertical) in the master branch of FF firmware (when the special protocol is used), that would be already sufficient. And maybe later it could be made configurable, but there's no rush. Thanks.

keirf, 2022-09-14:
Yes, but I'm not intending to change the default.

Wierzbowsky, 2022-09-14:
Then making it configurable via FF.CFG would be best. I will wait for the solution. 

With the screenless Gotek and the Blue Pill, it's essential to have the enhanced protocol enabled in order to configure the settings via Gotek's buttons. At the same time, the large size of the OSD makes it unmovable to other parts of the screen. As the upper left corner of the MSX screen is usually quite "busy", moving the OSD to either of the right corners would be the best idea, but with the large size OSD this is simply impossible - it occupies too large part of the MSX screen (which is 265x212). I hope that you understand the problem that we have here.

keirf, 2022-09-14:
Yes, I see...

Wierzbowsky, 2022-09-24:
I've made the boards to connect Blue Pill to all 3 color channels. Thanks to @keirf for the idea.

[image]

keirf, 2022-10-09:
Here is a new build for you to try, with a new config option. Try `osd-columns = 16`

You will need to be logged into GitHub for this download link to work:
https://github.com/keirf/flashfloppy/suites/8685546132/artifacts/391637682

Wierzbowsky, 2022-10-09:
Many thanks! Works as desired now with the following settings:

osd-display-order = 0,1
osd-columns = 16
display-off-secs = 5

Will you include this feature into one of the upcoming 3.xx releases? I will need to write an article with modding instructions and recommended settings in the FF.CFG file and then share it with the MSX community.

keirf, 2022-10-09:
I am fixing SD card support on 435 chip. Then I will do a release. Maybe in about a week.

Wierzbowsky, 2022-10-22:
Confirmed to be working with v3.36! Many thanks!

closed 2022-10-22 by Wierzbowsky