keirf/flashfloppy-osd #50: Blinker test with proper baud rating needed (currently 3000000)

Wierzbowsky, 2022-10-14:
During the FF OSD porting to MSX the Blinker test (blinky-test-f1-1.2.hex) had to output from Blue pill. Then @keirf said:

No blinking led is bad. No serial is weird. I may have broken the build slightly and it's not currently at 115200 baud. Actually I think that's likely. Try 3000000 baud (three million).

Would it be possible to release the fixed Blinky test please? Thanks. It's essential to identify the Blue Pill boards that will work with Goteks.

_Originally posted by @keirf in https://github.com/keirf/flashfloppy-osd/issues/48#issuecomment-1208625862_

keirf, 2022-10-15:
A correct blinky will be included in older greaseweazle releases. Try downloading, say, v0.20 of greaseweazle host tools (firmware and host tools used to be in the same download file). If that works you could link to that older release in the docs you are writing.

I will update blinky in latest greaseweazle firmware when I next make a release. But that's not happening immediately for firmware.

I could update the wiki to point at v0.20 release temporarily for blinky serial though?

Wierzbowsky, 2022-10-15:
Will try, thanks.

Wierzbowsky, 2022-10-16:
The 0.20 version worked well, thanks. I will reference this version in the FF OSD for MSX adaptation document.