keirf/flashfloppy-osd #57: FF OSD problem with arrow keys selecting disk images

ptyku, 2024-09-09:
The problem is with selecting a floppy disk image using the right and left arrow keys. When I try to select the next image from the catalog, instead of the next one (+1), it jumps exactly 5-6 images away. 

The problem is only when selecting floppy disk images. Using the right/left arrow keys in the settings menu causes no problems. 

The device was tested on two different Amigas and using different keyboards (it works identically). 

FF 3.42 and FF OSD 1.9 software installed

https://photos.app.goo.gl/ah7W553ifdcDk4vbA

keirf, 2024-09-09:
Which Amigas and which keyboards has it specifically been tested on (eg are they standard keyboards or something else)?

I have a vanilla 500 board and keyboard I can quite easily test on, if that's in your problematic set.

ptyku, 2024-09-09:
Both are A500 rev. 5 and 6a with standard mitrumi keyboard. I forgot to mention that it's an internal amigotek by zaxon, so two in one. Both stm32 chips on one board so nothing got unplugged between ff and ff osd. https://photos.app.goo.gl/2o4UU6CZ8ihCtYHY6Krzysztof Szopa
-------- Oryginalna wiadomość --------Od: Keir Fraser ***@***.***> Data: 9.09.2024  12:07  (GMT+01:00) Do: keirf/flashfloppy-osd ***@***.***> DW: ptyku ***@***.***>, Author ***@***.***> Temat: Re: [keirf/flashfloppy-osd] FF OSD problem with arrow keys selecting disk images (Issue #57) 
Which Amigas and which keyboards has it specifically been tested on (eg are they standard keyboards or something else)?
I have a vanilla 500 board and keyboard I can quite easily test on, if that's in your problematic set.

keirf, 2024-09-09:
Ok thanks I will investigate.

keirf, 2024-09-09:
Can you also tell me what chip is used for the Gotek side of the Amigotek? If you can't read the chip, you can use "MCU Info" from the FlashFloppy main menu (accessible when no USB drive is inserted).

ptyku, 2024-09-09:
MCU INFO: STM32F105 72Mhz 64kbKrzysztof Szopa
-------- Oryginalna wiadomość --------Od: Keir Fraser ***@***.***> Data: 9.09.2024  14:57  (GMT+01:00) Do: keirf/flashfloppy-osd ***@***.***> DW: ptyku ***@***.***>, Author ***@***.***> Temat: Re: [keirf/flashfloppy-osd] FF OSD problem with arrow keys selecting disk images (Issue #57) 
Can you also tell me what chip is used for the Gotek side of the Amigotek? If you can't read the chip, you can use "MCU Info" from the FlashFloppy main menu (accessible when no USB drive is inserted).

ptyku, 2024-09-09:
Picture of the chipKrzysztof Szopa
-------- Oryginalna wiadomość --------Od: Keir Fraser ***@***.***> Data: 9.09.2024  14:57  (GMT+01:00) Do: keirf/flashfloppy-osd ***@***.***> DW: ptyku ***@***.***>, Author ***@***.***> Temat: Re: [keirf/flashfloppy-osd] FF OSD problem with arrow keys selecting disk images (Issue #57) 
Can you also tell me what chip is used for the Gotek side of the Amigotek? If you can't read the chip, you can use "MCU Info" from the FlashFloppy main menu (accessible when no USB drive is inserted).

keirf, 2024-09-09:
Attachments including pictures get stripped from emails. You have to attach such things in the GitHub web interface.

keirf, 2024-09-09:
I assume then that this is old enough that also the OSD chip is an STM32F103?

ptyku, 2024-09-09:
Yes, STM32F103Krzysztof Szopa
-------- Oryginalna wiadomość --------Od: Keir Fraser ***@***.***> Data: 9.09.2024  15:55  (GMT+01:00) Do: keirf/flashfloppy-osd ***@***.***> DW: ptyku ***@***.***>, Author ***@***.***> Temat: Re: [keirf/flashfloppy-osd] FF OSD problem with arrow keys selecting disk images (Issue #57) 
I assume then that this is old enough that also the OSD chip is an STM32F103?