keirf/flashfloppy-osd #60: Issues with latest gcc-arm-none-eabi

Aergan, 2025-02-11:
Under Ubuntu 24.04.02 LTS with gcc-arm-none-eabi (15:13.2.rel1-2), the source will not compile.
`CC amiga.o
CC build_info.o
CC cancellation.o
CC config.o
CC console.o
CC i2c.o
CC main.o
main.c: In function 'main':
main.c:863:15: error: comparison between two arrays [-Werror=array-compare]
  863 |     if (_sdat != _ldat)
      |               ^~
main.c:863:15: note: use '&_sdat[0] != &_ldat[0]' to compare the addresses
cc1: all warnings being treated as errors`

Overriding the array-compare warning produces a bin with ignores hotkeys other than basic navigation and selection (arrows). E.g. Show/Hide OSD and Keyboard held no longer activate.

Under Ubuntu 19.10 with gcc-arm-none-eabi (15:7-2018-q2-6), the binary produced compiles and works as expected:
`CC amiga.o
CC build_info.o
CC cancellation.o
CC config.o
CC console.o
CC i2c.o
CC main.o
CC string.o
CC stm32f10x.o
CC time.o
CC timer.o
CC util.o
AS vectors.o
CPP FF_OSD.ld
LD FF_OSD.elf
chmod a-x FF_OSD.elf
OBJCOPY FF_OSD.bin
chmod a-x FF_OSD.bin
OBJCOPY FF_OSD.hex
chmod a-x FF_OSD.hex`

keirf, 2025-02-11:
Thanks for the report. The build failure is now fixed. I have a possible fix for the hotkey problem too, as it's probably due to not re-reading an interrupt-modified global variable in the main loop. Can you checkout branch `issue-60` and give it a try?

Aergan, 2025-02-11:
Thanks for the quick response!

It compiles ok now but the behaviour is the same as before (either ignoring the warning or your current change) sadly.
CTRL + ALT + RETURN: No response
CTRL + ALT + RIGHT: Brings up the OSD as expected, can navigate and select with LEFT, RIGHT, UP
CTRL + ALT + RETURN:: (With the OSD up) causes the OSD to flash but no other response. Keyboard held is not activated

keirf, 2025-02-11:
If the latest fixes in `master` don't do the job I'll have to whip out my test board and test it myself :o

Aergan, 2025-02-11:
Hi there, yes that has resolved the issue from what I can tell - thank you very much :)

All hotkeys are working correctly when compiled on gcc-arm-none-eabi (15:13.2.rel1-2)

keirf, 2025-02-11:
Thanks for testing! Phew, I can leave the test board at rest. There's definitely some fast and loose use of variables modified in IRQ handlers, so this may get revisited in a future GCC version!

closed 2025-02-11 by keirf