keirf/flashfloppy-osd #61: ff f9 key codes occur after every key press in Amiga Test Kit

shan2752, 2025-03-15:
Hello,

I noticed a lack of keyboard response in on Application after I recently installed a new GoEx Drive w/ Pills and made all the proper OSD connections with pins.

Testing the keyboard in Amiga Test Kit, every keypress includes an ff f9 raw key code and there is a delay between keypresses.  When I unhook the OSD keyboard headers, the issue is resolved.

Any idea what the issue is?  Thanks!

keirf, 2025-03-16:
This means that the keyboard and Amiga are out of sync. The FF keycode is probably re-sync bits getting clocked out at ~8Hz, and then when things are back in sync, the keyboard sends a "garbled keycode" message (that's F9).

Well all sorts of things could cause this. For a start, FF-OSD inserts its own handshakes because not all games and demos handle keyboard at all. But FF-OSD could be out of sync with the Amiga! But noone else has seen that... Or it could be a simple electrical problem -- somehow the OSD is adding too much capacitance or extra pull-up to the serial lines. Or noise.

What Amiga exactly is this? And what Kickstart version? Amiga Test Kit is interesting here. Do *all* key presses and releases have this problem, or only some? Can you report behaviour of pressing and releasing the tilde key (that's the one to the left of the '1' key) repeatedly, at a slow rate (press, wait for keycodes, release, wait for keycodes) -- the exact codes observed by ATK are interesting.

Other useful debug aids here may be the ability to re-flash the OSD firmware, and get serial logs from the OSD firmware. I don't know if you have these capabilities (nor whether the serial data is available on a pin on the GoEx drive).

shan2752, 2025-03-16:
Thanks for the quick reply!  This is on an Amiga 500, Rev 6A motherboard, with a 1.3 ROM.

When I press the tilde key once, I get 00 80 ff f9

When I hold the tilde key it displays 00 and then 80 ff f9 once I release it.

Pressing the tilde key multiple times I get 00 80 ff f9 80 80 80 80 80 80 80 80 ff f9 80 80, and the key only shows one keypress visually like it’s frozen.

Every key behaves the same with the ff f9 code, except the caps lock key, which displays a 62 when pressed to activate followed by a e2 ff f9 to deactivate.

This is with a new Kipper 2K keyboard that was sold by Amiga By the Lake.

So I just used an original keyboard, and the issue is gone!  The problem is being caused by the new keyboard.  With new keyboards being made, there should probably be some coordination here as a new keyboard will break FF OSD.

keirf, 2025-03-16:
Yeah the Amiga, OSD, and host all need to be in sync and that is not somehow the case with the AotL keyboard. Damn. And are these the legendary hens-teeth unobtainium keyboards which people forked out for and waited a lifetime to receive? I don't fancy my chances of getting one for testing if so!

keirf, 2025-03-16:
I suspect the handshake timing produced by FF OSD is not liked by the keyboard. It probably expects to see a high-to-low edge on the KDAT line, but if the last bit is low then there is no edge.

That's a theory, anyway. I would be happy to produce you a new firmware to test this theory out. If I did, would you be able to program it onto the Goex? It will certainly need a jumper and a usb-serial programmer.

keirf, 2025-03-16:
Cheers, it will be interesting to find whether Goex is "user serviceable" at all.

As for coordination, I'm happy to support new hardware if I am given access. Any of these new keyboards I'm sure I could tweak the firmware to support within an hour or so. However I make no money on FF-OSD at all so I have no plans to pay out of pocket to support non-original hardware, as I'm sure you can understand. The Amiga Store keyboard is A1200 only I believe, and thus is trivially compatible because A600 and A1200 have the keyboard controller on the motherboard, not as part of the keyboard.