keirf/flashfloppy-osd #62: Can't configure FF ODS via Putty

Wierzbowsky, 2025-04-04:
I was installing the FF OSD into MSX computer and I wanted to configure some options via serial interface. Using Putty with all recommended settings gets me to the configuration menu, pressing Reset on the Blue Pill gets me the current config. But when I try to send a command to the device it doesn't react. I tried with 2 different previously programmed boards and the result is always the same - no reaction. I remember a few years back I could program the device before installing it. No idea what I am doing wrongly now. Any help please?

[image]

keirf, 2025-04-04:
The fact that putty has echoed the key presses is not a good sign. And with newlines too? You want settings akin to "stty raw -echo"

Wierzbowsky, 2025-04-04:
No matter what settings are try both boards that I programmed a couple of years ago do not react to keypresses. Could you please let me know what character set is expected? By default it's UTF-8 in Putty. Also, is there a video how to configure this board? Maybe I am missing something, thanks.

Wierzbowsky, 2025-04-04:
I solved the problem by doing the config on the other machine with older UART driver.

closed 2025-04-04 by Wierzbowsky